frankarobotics/franka_ros #348: invaild message type

1358467720, 2023-06-15:
when i run cmd:
`rostopic pub --once /franka_gripper/move/goal franka_gripper/MoveActionGoal "goal: { width: 0.08, speed: 0.1 }"
`
i got :
ERROR: invalid message type: franka_gripper/MoveActionGoal.
If this is a valid message type, perhaps you need to type 'rosmake franka_gripper'

marcbone, 2023-06-15:
if you install franka_ros via apt  you shouldnt need to to anything. If you build franka_ros from source you only need to run catkin_make or catkin build and then source your devel/setup.bash.

1358467720, 2023-06-15:
> if you install franka_ros via apt you shouldnt need to to anything. If you build franka_ros from source you only need to run catkin_make or catkin build and then source your devel/setup.bash.

it really helpful. And  I have anthor question. why I can not see the franka_gripper/MoveActionGoal described in source code ?