frankarobotics/franka_ros #350: No publisher to /dual_arm/panda_1/franka_gripper/joint_states using franka_combined_control.launch

FSIE-group, 2023-06-26:
Hello, I am trying to config double franka panda arm in ROS using franka_combined_control.launch
I bringup the franka panda robots using`roslaunch franka_control franka_combined_control.launch robot_ips:="{panda_1/robot_ip: 192.168.1.10, panda_2/robot_ip: 192.168.1.12}"`. After bringup, I can get the topic using `rostopic list` and can also echo the topic using `rostopic echo /dual_arm/joint_states`. 
However, the joint position information about gripper in the topic  /dual_arm/joint_states is always 0. See as below.
[image]
Also, i found there is no publisher to /dual_arm/panda_1/franka_gripper/joint_states:
[image]
My franka_combined_control.launch file is as follows:
`<?xml version="1.0" ?>
<launch>

  
  <arg name="robot" default="$(find franka_description)/robots/dual_panda/dual_panda_example.urdf.xacro" />

  
  <arg name="args" default="" />

  
  <arg name="robot_id" default="dual_arm" />

  
  <arg name="hw_config_file" default="$(find franka_control)/config/franka_combined_control_node.yaml"/>

  
  
  <arg name="robot_ips"  default="{panda_1/robot_ip: 192.168.1.10, panda_2/robot_ip: 192.168.1.12}"/>

  
  <arg name="controllers_file" default="$(find franka_control)/config/default_combined_controllers.yaml" />

  
  <arg name="controllers_to_start" default="panda_1_state_controller panda_2_state_controller"/>

  
  <arg name="joint_states_source_list" default="[panda_1_state_controller/joint_states, panda_2_state_controller/joint_states, panda_1/franka_gripper/joint_states, panda_2/franka_gripper/joint_states]"/>
  
  <arg name="joint_names_1" default="[panda_1_finger_joint1, panda_1_finger_joint2]" />
  <node name="franka_gripper_1" pkg="franka_gripper" type="franka_gripper_node" output="screen">
    <param name="robot_ip" value="192.168.1.10"/>
    <param name="stop_at_shutdown" value="false" />
    <rosparam command="load" file="$(find franka_gripper)/config/franka_gripper_node.yaml" />
    <rosparam param="joint_names" subst_value="true">$(arg joint_names_1)</rosparam>
  </node>

  <arg name="joint_names_2" default="[panda_2_finger_joint1, panda_2_finger_joint2]" />
  <node name="franka_gripper_2" pkg="franka_gripper" type="franka_gripper_node" output="screen">
    <param name="robot_ip" value="192.168.1.12"/>
    <param name="stop_at_shutdown" value="false" />
    <rosparam command="load" file="$(find franka_gripper)/config/franka_gripper_node.yaml" />
    <rosparam param="joint_names" subst_value="true">$(arg joint_names_2)</rosparam>
  </node>

  <node name="$(arg robot_id)" pkg="franka_control" type="franka_combined_control_node" output="screen" required="true" >
    <rosparam command="load" file="$(arg hw_config_file)" />
    
    <rosparam subst_value="True">$(arg robot_ips)</rosparam>
    <param name="robot_description" command="xacro $(arg robot) $(arg args)" />
  </node>

  <group ns="$(arg robot_id)">

    <rosparam command="load" file="$(arg controllers_file)" />
    <node name="hw_controller_spawner" pkg="controller_manager" type="spawner" respawn="false" output="screen"
      args="$(arg controllers_to_start)"/>

    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen"/>
    <node name="joint_state_publisher" type="joint_state_publisher" pkg="joint_state_publisher" output="screen">
      <param name="robot_description" command="xacro $(arg robot) $(arg args)" />
      <rosparam param="source_list"  subst_value="true">$(arg joint_states_source_list)</rosparam>
      
      
      <param name="rate" value="30"/>
    </node>

  </group>

</launch>
`
In addition, when i use `roslaunch franka_control franka_control.launch robot_ip:=192.168.1.10` to bringup single arm, all the things work well.

How can i solve this problem.
Any suggestion would mean a lot!

marcbone, 2023-07-03:
I think the problem is that you did not set arm_id in the gripper_node. Compare to the [teleop_gripper.launch](https://github.com/frankaemika/franka_ros/blob/develop/franka_example_controllers/launch/teleop_gripper.launch).
You can also check the [teleop_joint_pd_examle_controller.launch](https://github.com/frankaemika/franka_ros/blob/develop/franka_example_controllers/launch/teleop_joint_pd_example_controller.launch). Maybe this example already is what you are trying to achieve. And even if not it should help you to find the problem in your launch file. Keep in mind that this launch file is using the franka_combined_control.launch, so you need to revert the changes that you made there to run the example.

mitsui29, 2024-02-28:
@aodongdong hi, i wanna start with the dual arm on the simulation and real panda. Did you try it on the gazebo? I'm looking for your sharing. Thanks a lot!

mikelasa, 2024-04-11:
Same as @mitsui29, I want to create simulation on gazebo for two panda robots and the dual impedance cartesian controller. I'm facing lots of issues though... Have you tried?