frankarobotics/franka_ros #353: Connect franka ros with Movegroup

saivojjala, 2023-08-03:
Can someone please help me connect franka ros with movegroup. I would like to move my franka emika panda robot using the moveit plugin on rviz

marcbone, 2023-08-04:
use the [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config) and run
 `roslaunch panda_moveit_config franka_control.launch robot_ip:=<ip-of-your-robot>`

closed 2023-08-04 by marcbone

saivojjala, 2023-08-06:
Does this interface require a real-time kernel? Is it at all possible to use the library with moveit without installing a different kernel. I tried installing the kernel but I get an error reading out of memory when i restart my laptop

888xc888, 2023-10-09:
> 使用[panda_moveit_config](https://github.com/ros-planning/panda_moveit_config)并运行 `roslaunch panda_moveit_config franka_control.launch robot_ip:=<ip-of-your-robot>`
Hi I'm using fr3, how can I modify panda_moveit_config to connect the moveit to the real robot arm? thank you!