frankarobotics/franka_ros #355: publish hz of /franka_state_controller/joint_states does not reach 1000

sm3304love, 2023-08-15:
When using the franka_ros package, I get only 650hz despite setting the publish rate of /franka_state_controller/joint_states to 1000. How should I solve this?

[image]

franka_ros version : 0.8.0
libfranka version : 0.8.0

undefinedcodezhong, 2023-08-16:
Do you set up the real-time kernel on your device? 
https://frankaemika.github.io/docs/installation_linux.html?highlight=real+time#setting-up-the-real-time-kernel

sm3304love, 2023-08-16:
> Do you set up the real-time kernel on your device? https://frankaemika.github.io/docs/installation_linux.html?highlight=real+time#setting-up-the-real-time-kernel

Yes, I already installed real time kernel on ubuntu 18.04. The robot itself is good at moving. My concern is that the controller's control cycle is 1000 Hz, but reading the point state is less than 1000 Hz

undefinedcodezhong, 2023-08-16:
Perhaps it is the performance loss of ROS in communication. I found a blog talking about the issue. 
https://zhuanlan.zhihu.com/p/624606278

marcbone, 2023-08-16:
I wouldnt worry about it too much. The controller itself runs at 1000 Hz. However, the state publisher is not meant to be realtime. If you want to access the state at 1000Hz best practice would be to write your own controller that uses the FrankaStateInterface. You can verify that the control is running stable by looking at the [control_command_success_rate](https://github.com/frankaemika/franka_ros/blob/358406952597c3735d43625595650901be84a5d8/franka_msgs/msg/FrankaState.msg#L45) in the franka state message. If it is above 0.95 everything is fine