frankarobotics/franka_ros #356: Panda research 3 PID gains tweaking

samdjExceenis23, 2023-08-21:
I'm using franka_ros and Moveit to control the robot. Whenever I increase the speed/acceleration via  Moveit move group, the robot starts to oscillate on some joints, and/or throws this error:
```
[ERROR] [1692376851.550258786]: libfranka: Move command aborted: motion aborted by reflex! ["joint_motion_generator_velocity_discontinuity", "joint_motion_generator_acceleration_discontinuity"]
control_command_success_rate: 0.66
```
I was wondering if it was possible to use a PID in order to stabilize the robot movement but I can't find where to add those parameters.
The ```default_controllers.yaml``` file is loaded and contains PID gains for the ```effort_joint_trajectory_controller``` but nothing on the ```position_joint_trajectory_controller```, is it where it is supposed to go ?
When using the robot at max speed on Franka desk, such oscillations never happen.

marcbone, 2023-08-21:
To me it seems like your problem is your control_command_success_rate. It is very low. You should run the communication test from libfranka and try to find the issue in you network setup

samdjExceenis23, 2023-08-21:
But does that explain the oscillation issue ?

marcbone, 2023-08-24:
With your success rate I wouldnt be surprised to get oscillation problems when the rate limiter is enabled. You definitely have to fix your communication issues to properly control the robot with FCI

samdjExceenis23, 2023-09-21:
I have fixed the communication problem, and I'm now getting this when launching the communication test :
#######################################################
Control command success rate of 10000 samples: 
Max: 1.00
Avg: 0.99
Min: 0.71
#######################################################

But still, the robot is oscillating when I set high values for speed and acceleration (usually an acceleration scaling higher than 0.5 and a velocity scaling higher than 0.7) though Rviz, but seem to work fine with low acceleration scaling values (anything less than 0.5).

It's worth noting that the oscillations come at the end of the movement.

marcbone, 2023-09-22:
This looks much better now. Try to disable the rate-limiter here: https://github.com/frankaemika/franka_ros/blob/develop/franka_control/config/franka_control_node.yaml#L14 . Then tell me if there are still oscillations