frankarobotics/franka_ros #373: fr3_moveit_config

888xc888, 2023-10-09:
I'm using fr3, how can I modify panda_moveit_config to connect the moveit to the real robot arm? thank you!
I'm a newbie and don't understand many of the concepts, can you suggest a more detailed modification?

BarisYazici, 2024-01-04:
FR3 robot is supported by ROS 2 and there is a moveit_config file that supports it. FR3 and Panda has similar kinematics you can use the panda moveit config  in the franka_ros2 repo. https://github.com/frankaemika/franka_ros2

utomm, 2025-04-10:
for those who are still sticking on ROS1, check [fr3_moveit_config](https://github.com/utomm/fr3_moveit_config)

winkma, 2025-04-11:
Thanks @utomm for pointing that out 😊 please keep in mind though that ROS noetic will reach its end of life in May 2025