frankarobotics/franka_ros #383: Link 'panda_hand_tcp' declared as part of a chain in group 'panda_manipulator' is not known to the URDF

upc-ghy, 2024-04-07:
PC
Ubuntu 20.04
ros noetic
I have correctly installed libFranka and Franka'ros according to the tutorial before installing Moveit.
When I follow https://moveit.ros.org/install/source/ When installing Moveit 1.0 without errors in the tutorial, I conducted the following tests on the installed Moveit:

﻿roslaunch franka_control franka_control.launch robot_ip:=172.16.3.69 load_gripper:=true

The following error was discovered as a result. My panda_arm are normal and can be planned and executed. Please give me some guidance, I would greatly appreciate it.

```
**[ERROR] [1712405841.172965565]: Link 'panda_hand_tcp' declared as part of a chain in group 'panda_manipulator' is not known to the URDF
[ WARN] [1712405841.173009900]: Group 'panda_manipulator' is empty.
[ERROR] [1712405841.173814276]: Link 'panda_hand_tcp' specified as parent for end effector 'panda_hand_tcp' is not known to the URDF**
[ WARN] [1712405841.173892827]: Link 'panda_link0_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.173920940]: Link 'panda_link1_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.173950750]: Link 'panda_link2_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.173976903]: Link 'panda_link3_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.173997879]: Link 'panda_link4_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.174017783]: Link 'panda_link5_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.174037317]: Link 'panda_link6_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.174056920]: Link 'panda_link7_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.174077081]: Link 'panda_hand_sc' is not known to URDF. Cannot specify collision default.
[ WARN] [1712405841.174107003]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174127863]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174148131]: Link 'panda_link0_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174168542]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174188269]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174208787]: Link 'panda_link1_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174228992]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174249053]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174269316]: Link 'panda_link2_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174289396]: Link 'panda_link3_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174311769]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174331848]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174351543]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1712405841.174371157]: Link 'panda_hand_sc' is not known to URDF. Cannot disable/enable collisons.
[ INFO] [1712405841.174673954]: Loading robot model 'panda'...
[ WARN] [1712405841.174971388]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1712405841.175016775]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
**[ WARN] [1712405841.175768763]: Group 'panda_manipulator' must have at least one valid joint
[ WARN] [1712405841.175795935]: Failed to add group 'panda_manipulator'
[ERROR] [1712405841.175964231]: Group state 'ready' specified for group 'panda_manipulator', but that group does not exist
[ERROR] [1712405841.175987836]: Group state 'extended' specified for group 'panda_manipulator', but that group does not exist**
```

husmanhameed, 2024-07-04:
Hi, I am facing the same issue. Have you found any fix for this?

upc-ghy, 2024-07-05:
@husmanhameed 
Hi, I have not found a solution for this problem so far.