frankarobotics/franka_ros #392: Franka Research 3 Low precision

samdjExceenis23, 2024-06-03:
Hello,

I'm working with a Franka Research 3, using Moveit Movegroup, and I've noticed that the robot's precision is really low, so for example, if I request a position and orientation, sometimes it offsets that goal position by ~ 1cm, which is high considering our application.

I tried to log the end effector's current pose, but the issue is that I'm getting only a small difference between iterations, a difference that don't reflect the current real pose. I also tried different type of movements (joint, Cartesian, a combination of both).

Using different IK solver, and going from KDL to TracIK and BioIK improved things, but there still is that offset of around 1cm.

Platform:  Ubuntu 20.04
Franka ROS: noetic-devel
libfranka: 0.10.0

felixog1, 2025-08-04:
same issue here, if I manually place my robot (Franka Emika Panda) twice at the same position but with a different orientation (rotated 180deg) i get a ~1cm error in the position reading

GBZH, 2025-10-17:
Hi, 
Currently I am facing the same issue. 
It has been more than one year since you have posted yours. Did anyone from Franka ever answer to you?