frankarobotics/franka_ros #395: Error: when trying to execute roslaunch panda_moveit_config franka_control.launch robot_ip:=<fci_ip>

rucha6198, 2024-07-03:
When I try to run the `roslaunch panda_moveit_config franka_control.launch robot_ip:=<fci_ip>` command, I get the following error.

`FrankaHW: Failed to initialize libfranka robot. libfranka: Connection error: Host not found [ERROR] [1720028829.142093723]: franka_control_node: Failed to initialize FrankaHW class. Shutting down!`

It shows 'You can start planning!' after this error but automatically shuts down everything. I am using ROS Noetic with Ubuntu 20.04. I have installed franka_ros. I also installed libfranka separately because I was not sure if the error was due to libfranka.

I would appreciate it if someone could help me out with this. 

Thank you!

yhnb0308, 2024-10-19:
The reason is that the real-time kernel is not properly configured
//1. Add real-time group license
sudo addgroup realtime
sudo usermod -a -G realtime $(whoami)
//2. Add restrictions to real-time groups
sudo gedit /etc/security/limits.conf
//3. Add the following restrictions to the opened file
//Restricted content
@realtime soft rtprio 99
@realtime soft priority 99
@realtime soft memlock 102400
@realtime hard rtprio 99
@realtime hard priority 99
@realtime hard memlock 102400

Then restart the real-time kernel

yhnb0308, 2024-10-19:
> When I try to run the `roslaunch panda_moveit_config franka_control.launch robot_ip:=<fci_ip>` command, I get the following error.
>
> `FrankaHW: Failed to initialize libfranka robot. libfranka: [...]

The reason is that the real-time kernel is not properly configured
//1. Add real-time group license
sudo addgroup realtime
sudo usermod -a -G realtime $(whoami)
//2. Add restrictions to real-time groups
sudo gedit /etc/security/limits.conf
//3. Add the following restrictions to the opened file
//Restricted content
@realtime soft rtprio 99
@realtime soft priority 99
@realtime soft memlock 102400
@realtime hard rtprio 99
@realtime hard priority 99
@realtime hard memlock 102400

Then restart the real-time kernel