frankarobotics/franka_ros #400: FrankaHWSim: how to lock joints in position

marcomasa, 2025-01-15:
I am trying to use the Franka in Gazebo.
The arm is mounted on a mobile robot platform.

I've managed to integrate the existing URDF models in my own robot model, 
but now when i move the mobile robot platform, the inertia is transferred to the arm and its joints move after the robot has experienced some acceleration (e.g. starting or stopping). `

See the attached gif for this:

[image]

Is there a way to fix the joints in position by e.g. calling some service or setting a target position?

winkma, 2025-01-16:
Hi Marco!

All of our controllers in some way use gravity compensation to control the motor torques. Since the robot assumes that the robot base only experiences an acceleration from the earth gravity, the compensation will be wrong once you accelerate the base -> that's probably why you're seeing the joints move.
What my best idea would be to keep the robot standing still:
- implement your custom joint impedance controller with a relatively high joint impedance
- just before updating the joint torques
  - add a torque to counter our internal gravity compensation
  - add another torque to compensate the effects of gravity + base acceleration
You can get the gravity torque vector from the model in franka_hw. Since you already have a working URDF and should know the base acceleration, it should be not too hard to use a dynamics library such as pinocchio or KDL to compute the real effects from the acceleration + gravity.

I hope that helps even though we don't have an out of the box solution for your use case yet!

marcomasa, 2025-01-28:
Hi @winkma,

thanks so much for your detailed response!
I do understand the issue now a bit better, however, the implementation seems a bit over-complicated for what i am trying to achieve. In my setup, i do not need to actually move the arm once its in that position. 

Do you think it would be possible to just change all the joint types to "fixed", rather than revolute?
While this is a pretty harsh workaround, it could help me achieve what i want to do.

winkma, 2025-02-20:
Hi @marcomasa ,
that would probably work :) I am not sure though how many things you would need to change as a consequence, there may be errors to be resolved afterwards.