frankarobotics/franka_ros #404: Non-Zero Values from O_F_ext_hat_K Without Applied External Force

bdlim99, 2025-02-24:
**Description:**
I'm observing non-zero values from the `O_F_ext_hat_K` topic even when no external force is applied to the robot's end effector (EE) and no accessories are attached.

**Steps to Reproduce:**
1. Launch the Franka controller using:
   ```
   roslaunch franka_control franka_control.launch
   ```
2. In another terminal, run:
   ```
   rostopic echo franka_state_controller/franka_states | grep O_F_ext
   ```
3. Observe output similar to:
   ```
   O_F_ext_hat_K: [1.4046984925349721, -0.3746988189998095, 1.7208226415931065, 0.0025019231454423, 0.4513628440880607, -0.1473264999079312]
   O_F_ext_hat_K: [1.3852304678927174, -0.36299994795652923, 1.7134695277137726, -0.006568094169452518, 0.4395951988301791, -0.14457598724463322]
   O_F_ext_hat_K: [1.3910224549444952, -0.36906682006852587, 1.7129011732986914, -0.002232087659523152, 0.4441428239831187, -0.14767820703593915]
   ...
   ```

**Observations:**
- With the robot held in a fixed pose, the `O_F_ext_hat_K` value remains constant, suggesting that these readings are not merely random noise.
- When changing the robot's pose and then removing any external forces, the reported `O_F_ext_hat_K` values change accordingly, indicating a dependency on the robot's configuration.

**Questions:**
1. Is `O_F_ext_hat_K` intended to directly represent the external wrench acting on the robot's EE? If so, why are the values non-zero when no external force is applied?
2. Why does the value of `O_F_ext_hat_K` vary with the robot's pose, even in the absence of external disturbances?

Any clarification on the calibration process or the underlying estimation method that might explain this behavior would be greatly appreciated.

winkma, 2025-02-24:
Hi @bdlim99 ,

The estimated external wrench `O_F_ext_hat_K` signal is calculated from the joint torque sensor measurements `tau_J` , using the robot model to compute the estimated external joint torques `tau_ext_hat_filtered`. Although mechanical wear can be a factor (e.g. friction), the biggest measurement errors here come from "torque sensor offsets". Those were calibrated in our production line just before you received the robot, but they are known to change over time, especially when there have been impacts to the torque sensors.
From the `tau_ext_hat_filtered` you can use the kinematics (aka the Jacobian) to compute `O_F_ext_hat_K`. That could be the reason why the error changes over time as in some configurations those joint torque offsets could be amplified or joint offsets in different joints could also cancel each other out.

I hope this was at least helpful to understand the issue, even though we cannot fix it for you right now 😒 

If you have a Franka Research 3, please stay tuned for updates, as we will release a feature in Desk for users to re-calibrate the offsets! 😸

shr-eyas, 2025-06-26:
@winkma Is that method to re-calibrate been introduced yet? 
We have FR3 and would love to try it out!

winkma, 2025-06-27:
Hey @shr-eyas , thanks for following up 😊 Yes, the torque calibration feature has been released with system image 5.8.0 🥳 Please try it out, and we'd love to hear your feedback on it if you have any!

Tiny-Dog-668, 2026-05-05:
Hi, thanks for the reply.  @winkma

I have updated to system image 5.8.0 and executed the torque sensor calibration routine. The calibration was performed with the end-effector removed and the robot configured as `no end-effector`.

However, after calibration, the estimated external wrench still seems to be strongly affected by the robot pose and motion state. More specifically:

1. In a static no-contact condition, simply changing the end-effector pose still causes a significant change in `O_F_ext_hat_K`;
2. During slow no-contact Cartesian motion, the estimated external force/torque also shows noticeable fluctuations;
3. The magnitude of these variations can be close to, or even larger than, the threshold I would like to use for contact detection, which makes it difficult to distinguish real contact from pose/motion-induced bias.

Is this expected behavior for the current external wrench estimation?
If I want to use the built-in Franka external wrench signal for contact detection, do I still need to implement an additional pose-dependent baseline compensation / bias compensation?
Or is there any officially recommended signal, filtering method, or calibration procedure to reduce this pose-dependent error?

Thanks!

undefeatedsoul, 2026-10-29:
Hi，@Tiny-Dog-668 ，I would like to ask if you have found a solution to the problem where the estimated force changes due to the posture?